keebio/iris-case #3: How long are all the paths in your design combined?

danielo515, 2018-01-14:
Hello,

I wanted to order this case from lasergist because it is sold out on keeb.io. Thanks to the README I can fulfill the creation form without problems except for one field, the PATH LENGTH:

>PATH LENGTH
>How long are all the paths in your design combined?

If you can tell me what is the correct value for that parameter I'll ve very grateful

MungFuSensei, 2018-01-15:
Would love an answer for this as well.

nooges, 2018-01-15:
For the switch plate, path length is 2224.7mm. For the bottom plate, it's 626.4mm.

However, I probably wouldn't order from Lasergist, as they seem to have problems filling orders or communicating these days: https://www.reddit.com/r/MechanicalKeyboards/comments/73ub8z/r_lasergist_no_longer_active_but_website_still/

closed 2018-01-15 by nooges

danielo515, 2018-01-16:
Thank you very much @nooges !! 
That information should be added to the readme in my opinion.

I made an order to lasergist this summer (one or maybe two months ago) and I had no problem at all. Maybe that is a matter of the past ?

thorinmesser, 2018-12-18:
+1 for putting this in the readme.