keebio/iris-case #5: Ponoko stroke width?

gwerbin, 2018-01-27:
Ponoko's laser cuts a 0.01mm stroke. Do the design files need to be adjusted for that?

geoah, 2018-02-19:
@gwerbin did you try to get these cut with ponoko? Wondering how they came out.

gwerbin, 2018-02-19:
I didn't get these plates cut but I did get a custom design (via Swillkb) done by Ponoko. I didn't set a kerf in Swillkb, and the switches fit pretty well.

I'm not sure if that helps at all. Maybe I can compare the switch openings in the Iris file to the ones in my file and adjust accordingly.

But now that you've teased the Rorschach maybe I'll put this whole idea on hold ;)

nooges, 2018-02-22:
Yeah, when generating files for Ponoko, you need to set the color of the lines to blue and set stroke width to 0.01mm. It's not really the cutting width of their laser, it's just the format that they'd like the files to be in.

closed 2018-02-22 by nooges