keebio/iris-case #20: REQUEST:

jocelynmallon, 2019-07-22:


nooges, 2019-07-23:
Yeah, I'll need to do a redesign in a different CAD program sometime, because the one I did the STL files in doesn't have an option to export higher quality output.

si14, 2020-04-02:
@nooges would you be interested in Fusion360 files for rev4? Aside from poly count which is very visible on PLA printed curves, I've also noticed a few other problems in current 3d printed mids:

* they can be substantially lower for rev4 if used with inserts instead of standouts
* usb-c port interferes with the mid layer and makes it slightly off centre
* tolerances on holes for nuts to hold tenting legs can be tighter to hold nuts in place
* round openings ideally should be teardrop shaped, otherwise their tops are flat, not round

I can try and fix all that, but knowing it'll be helpful for someone else would make it more worthwhile

nooges, 2020-04-02:
@si14 Yeah, I would be interested, as I don't have time to redo it all myself.

si14, 2020-04-02:
@nooges I'll try to come up with something this weekend and let you know how it works out. Do you have any constraints or preferences on hardware? How dead set are you on using the standoffs? Any tips or advice if you don't mind me asking?

si14, 2020-04-14:
@nooges sorry for a delay. I think I got it more or less right https://a360.co/2Xz4Tp7 (can you download it from there?)

Here are a few changes:

[image]

Current STL doesn't quite fit and bulges out, new one doesn't. I think USB-C port is a touch too close in the current one

[image]

USB-C cutout is a bit more square so it prints better. Also no holes on the sides

[image]

TRRS cutout is aligned with the bottom plate and the hole for the socket is now teardrop shaped so it doesn't interfere with the socket

[image]

It's really hard to show on the photos, but higher poly STLs are visibly smoother on the curves

I've also made standouts fit tighter so it's easier to screw the plates. They were very loose with the current STL, making it difficult to screw everything in one go.

I've done my best to keep it as parametric as I could, so minor adjustments shouldn't be a problem. The only hacky part is finding centres for those holes using constraints, as they exported as curves and Fusion can't tell they are circles with a centre point. Please let me know if I can help in any way.

Here it is as an STL just in case: 
[mid_layer.stl.zip](https://github.com/keebio/iris-case/files/4476188/mid_layer.stl.zip)

I also think that if you will ever want to mass-produce them for sale it can be optimised quite a bit. I'd reduce thickness where it's not needed and cut them in two so that multiple mid layers can fit on a bed.

I think for my own keyboard I'll probably ditch the bottom plate altogether and design a bottom with embedded nuts so it's slimmer and to avoid bottom screw heads scratching my table. I can share a design if you're interested.

Oh yes, and it'd be great if you could provide a dxf version for `iris-plate-design-rev2-and-rev3.ai` going further. I had to install a trial version of Illustrator just to convert it to something Fusion will import, and Inkscape doesn't handle layer names.

austenc, 2020-10-13:
Hey @si14 

These look awesome! I'd like to make one that supports ergodox legs for tenting. Unfortunately I can't seem to download the file from the link you posted. Any chance you could attach a zip of those here, or email them to me? 

Thank you!

si14, 2020-10-14:
Hey @austenc ! Thanks for your kind words. Here it is: 
[mid layer v9.f3d.zip](https://github.com/keebio/iris-case/files/5379990/mid.layer.v9.f3d.zip)

Please let me know if it works alright for you (and I hope you won't be too offended by my modelling…)

austenc, 2020-10-15:
Thank you very much @si14 -- the modeling is great by the way! Saves me a bunch of trial and error trying to get the fit right myself. I printed one from the STL files above and it fits very well, thank you!

I haven't added the ergodox stuff yet, but I did export a left and right side from your fusion file in this zip archive. Perhaps others who find this thread will find it helpful. 
[iris-rev4-midlayers.zip](https://github.com/keebio/iris-case/files/5381515/iris-rev4-midlayers.zip)

nooges, 2020-10-27:
@si14 @austenc I've added the new mid layer files in this commit: af8f4875c47165192e01a78bc5c48c85ca2b0284

I'll keep this issue opened, as I still need need to get around to adding the Ergodox leg version of the updated design.

austenc, 2020-10-28:
Awesome @nooges, and thanks for the epic keebs 🤗

nooges, 2020-10-28:
@austenc Added Ergodox leg version in today: https://github.com/keebio/iris-case/blob/master/rev3-and-rev4/Iris%20R4%20Middle%20Layer%20-%20Ergodox%20Tent%20Kit%20for%201.5mm%20Plate.stl

closed 2020-10-28 by nooges