keebio/iris-case #21: Issue with the 3-D Case

CommonSenseUsed, 2019-07-23:
@CharlesLenk whenever I render the model I get no top-level geometry to render. Is there a fix for this? I changed the user configurable options to 

`two_unit_thumb = false;`
`stabilizers = true;`
`usb_c_pro_micro = false;`
`trrs_diameter = 6.55;`
`selector = 5;`

CharlesLenk, 2019-07-24:
You're right! Sorry about that, there's a missing call to draw the case. Adding the following call after the user defined options will fix the issue: `drawCase();`

I'll get a PR in to correct this as well.

CharlesLenk, 2019-07-24:
Ah, I see you already found it. Thanks for submitting the change! I don't have merge permissions, but I've added an approval for what it counts.

nooges, 2019-07-24:
Fixed with PR

closed 2019-07-24 by nooges