keebio/iris-case #29: No middle layer can be resized to 153.7mm x 132.5mm

jjkaram, 2020-06-17:
Hi Maintainers,

Thank you for your work on this project! Quick question, it seems to be that there is no middle plate for rev4 that can be resized to the appropriate dimensions specified in the readme file. I am missing something?

Thank you

lalde, 2020-06-28:
I had the same issue, I couldn’t resize it to the proper dimensions when importing the file on Ponoko or Sculpteo websites. I ended up taking the .ai file and modifying the dimensions directly in Adobe Illustrator. I then got the middle layer cut by Ponoko and it looks great. Hope this can help :)

nooges, 2020-11-19:
Ignore the README file dimensions for the mid, that was for one specific version of it.

closed 2020-11-19 by nooges