keebio/iris-case #32: Switch plate hole size

ThreepE0, 2021-03-29:
Hello, hope all is well.  I'm noticing when I open the illustrator file for 1.5mm thick switch plate in Illustrator, the holes in the plate measure 15.45mm wide (too wide) and 13.85mm tall (not tall enough.) Verified by printing out and trying to fit switches.  I created new illustrator files and converted to svg then stl, and once printed they seem to fit much better (though to keep things simple I went with plain square holes.)

I'm wondering if I'm missing something though, because I see others are mentioning that they've had success with these files.

Any help would be greatly appreciated

nooges, 2021-03-29:
The larger width is to support Alps-compatible switches, and the 13.85mm height was to support kerf for laser cutting, although even without kerf, 13.85mm still tends to work well.

ThreepE0, 2021-03-29:
hmm.... Would you be able to measure the inside of one of the resulting holes with calipers to see what the resulting cavity measures as on the steel and/or acrylic plates?  I'm having to bump up to 14.25mm in order to just barely fit (14.3 fits loosely,) though I'm 3d printing and not cutting.

I'm only printing in an effort to get the measurements right so I can send off plates to get cut though.  I would just use the provided design files, but I'm trying to add two thumb buttons per plate.

Thanks again

closed 2021-03-31 by ThreepE0