keebio/iris-case #41: PE Middle Layer USB

derpmcgerp1080p, 2022-06-23:
Hello,

The model for the PE middle layer (Ergodox/nolip) has a very small hole for the USB compared to the legacy edition. Because of this, it is incompatible with many of my USB cables as the connector is too large to fit in the whole. Requires heavy sanding to give enough clearance for cable to be inserted. Request that the main USB holes be widened to accommodate third party USB cables.7
[image]

nooges, 2022-07-05:
Can you measure the height and width of the housing of your USB-C cable so I can make the adjustments?

derpmcgerp1080p, 2022-07-28:
Absolutely.

Height=7.70mm
Width=13.00mm