keebio/iris-case #45: Stabilizers don't fit in PE Acrylic 3mm

konisteel, 2024-10-29:
Hey, so i laser cut the Iris PE Switch Plate Mx 3mm and after receiving it my stabilizers & the switches fall through the thumb area because the space it too large. The shop did not enlarge that area. How am i supposed to fit the stabilizers? I am going to handwire it so the stabs have to fit in the plate

https://github.com/user-attachments/assets/364dcd68-56ff-40d8-9d20-8b80f713b90c

 
[image]

Thanks for the help

nooges, 2024-10-29:
The cutout around the 2u key for 3mm plate is large to accommodate the stabilizer wire/housing. Otherwise, the wire and housing would hit the underside of the plate. Hence, the 3mm plate isn’t geared towards handwiring and more for use with a PCB.

konisteel, 2024-10-29:
Yea i figured that part out later after a lot of troubleshooting. Is it possible to use an adapter over there? The adapter will be 1.5mm and hold the switch & the stabs. Just a thought

nooges, 2024-10-29:
I’ll have to check my files, but I may have already designed a 3D-printable adapter to hold the switch in-place.

konisteel, 2024-10-29:
There is already a file to hold 2 1u switches but i want to use 1 2u
If you have the files, It would be a great help 🙏.
Thanks

konisteel, 2024-10-29:
Did you fine the stl file?

nooges, 2024-10-29:
I checked my Fusion 360 projects, and it doesn't look like I've designed a
2u adapter yet, but I can give it a go later this week.

konisteel, 2024-10-29:
That Would help me out a bunch, and others who will use it :D 
Can't wait for the files and the new update

nooges, 2024-10-31:
Yeah, I'm working something up right now and will do some test printing

konisteel, 2024-11-01:
Great!

konisteel, 2024-11-03:
Hey when can we expect the stl file :p

konisteel, 2024-11-04:
[keyboard adapter.zip](https://github.com/user-attachments/files/17617488/keyboard.adapter.zip)

nooges, 2024-11-06:
Turns out I did have something designed previously, see how this works out for you.
[Iris PE 2u Adapter for 3mm Plate.stl.zip](https://github.com/user-attachments/files/17649087/Iris.PE.2u.Adapter.for.3mm.Plate.stl.zip)

nooges, 2024-11-06:
If recutting the plate is an option, you could go with the older-style cutout in iris-PE-switch-plate-mx-alps.dxf

konisteel, 2024-11-06:
Hey how does the file work? I'm confused
[image]

nooges, 2024-11-06:
It’s not particularly great with handwiring though, so I don’t really have a good solution for using the plate as-is

image.png (view on web)

konisteel, 2024-11-06:
Alright i'll receive the 3d printed adapter that i uploaded and check if everything works great. I'll keep you updated but so far if you use the top plate than a little sanding is necessary or else it'll fit on the plate no problem with normal mx switches and cherry stabs